dorssel/usbipd-win #1239: usbipd failed to attach

minicarpet, 2025-08-01:
Probably again the same issue, I can't attach my device using usbipd.

My device is a Jetson Orin NX which is in recover state, the first step works and the device can be attached to send the boot (device 0955:7323  APX), but then the device restart as mass storage (1d6b:0104  Mass storage USB) and I got the issue.

```
usbipd.exe attach --busid 1-8 --wsl --auto-attach
usbipd: info: Using WSL distribution 'Ubuntu-24.04' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Detected networking mode 'nat'.
usbipd: info: Using IP address 172.24.16.1 to reach the host.
usbipd: info: Starting endless attach loop; press Ctrl+C to quit.
WSL Monitoring host 172.24.16.1 for BUSID: 1-8
WSL 2025-08-01 18:04:36 Device 1-8 is available. Attempting to attach...
WSL 2025-08-01 18:04:36 Failed to attach device 1-8
WSL 2025-08-01 18:04:54 Device 1-8 is available. Attempting to attach...
WSL 2025-08-01 18:04:54 Failed to attach device 1-8
WSL 2025-08-01 18:05:11 Device 1-8 is available. Attempting to attach...
WSL 2025-08-01 18:05:11 Failed to attach device 1-8
WSL 2025-08-01 18:05:29 Device 1-8 is available. Attempting to attach...
WSL 2025-08-01 18:05:29 Failed to attach device 1-8
WSL 2025-08-01 18:05:46 Device 1-8 is available. Attempting to attach...
WSL 2025-08-01 18:05:46 Failed to attach device 1-8
WSL 2025-08-01 18:06:04 Device 1-8 is available. Attempting to attach...
WSL 2025-08-01 18:06:04 Failed to attach device 1-8
WSL 2025-08-01 18:06:22 Device 1-8 is available. Attempting to attach...
WSL 2025-08-01 18:06:22 Failed to attach device 1-8
WSL 2025-08-01 18:06:39 Device 1-8 is available. Attempting to attach...
WSL 2025-08-01 18:06:39 Failed to attach device 1-8
WSL 2025-08-01 18:06:56 Device 1-8 is available. Attempting to attach...
```

On usbipd server:

```
usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = ''
dbug: Usbipd.Server[1000]
      usbipd:Port = 3240
dbug: Usbipd.Server[1000]
      5.1.0-41+Branch.master.Sha.fe06088bb6916e6722bd66b03d245141ae8ec6c1
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\logan.gallois
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.Server[1000]
      new connection from 172.24.16.71
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.24.16.71
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_DEVLIST
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.24.16.71
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId) + 0x153
         at Usbipd.VBoxUsb.<ClaimDeviceOnce>d__0.MoveNext() + 0x6b
      --- End of stack trace from previous location ---
         at Usbipd.VBoxUsb.<ClaimDevice>d__1.MoveNext() + 0x17e
      --- End of stack trace from previous location ---
         at Usbipd.VBoxUsb.<ClaimDevice>d__1.MoveNext() + 0x2b9
      --- End of stack trace from previous location ---
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__8.MoveNext() + 0xf03
      --- End of stack trace from previous location ---
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__8.MoveNext() + 0x18f9
      --- End of stack trace from previous location ---
         at Usbipd.ConnectedClient.<RunAsync>d__5.MoveNext() + 0x32f
dbug: Usbipd.Server[1000]
      connection close: Unable to find the specified file.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.24.16.71
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_DEVLIST
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.24.16.71
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId) + 0x153
         at Usbipd.VBoxUsb.<ClaimDeviceOnce>d__0.MoveNext() + 0x6b
      --- End of stack trace from previous location ---
         at Usbipd.VBoxUsb.<ClaimDevice>d__1.MoveNext() + 0x17e
      --- End of stack trace from previous location ---
         at Usbipd.VBoxUsb.<ClaimDevice>d__1.MoveNext() + 0x2b9
      --- End of stack trace from previous location ---
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__8.MoveNext() + 0xf03
      --- End of stack trace from previous location ---
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__8.MoveNext() + 0x18f9
      --- End of stack trace from previous location ---
         at Usbipd.ConnectedClient.<RunAsync>d__5.MoveNext() + 0x32f
```

Note that on windows I can't enumerate the device:

[image]

I have tried to rebuild Linux kernel for my WSL (see attached file).

dorssel, 2025-08-15:
Looks like your device and/or driver does not like dynamic reset. Try using `bind --force`.

dorssel, 2025-08-20:
The "device cannot start" could be the problem. But please try bind --force again. It should change the driver to US UP Shared Device. Or give an error if it cannot succeed.

minicarpet, 2025-08-21:
So what's next now ? do this mean I just can't use my device ?

minicarpet, 2025-08-27:
The device works well on Linux. So this is probably a problem with Windows...

jjaccard-Asulab, 2025-10-20:
Hello, 
I came across a similar issue where I could force share a device but not attach. The device was not listed as a USBIP Shared Device and the little warning triangle was next to it with the original name (Segger J-Link). The usbipd server log looked very much the same.

What worked for me was downgrading usbipd to 5.2.0 from 5.3.0. I have no idea why.

Let me know if any other log or information can help.

dorssel, 2025-10-20:
@jjaccard-Asulab 
Was the device force-bound? Did you first update from an older version to 5.3.0 before you downgraded? Or was it a fresh 5.3.0 install that did not work?

dorssel, 2025-10-23:
> When I do `usbipd bind -b=2-6 --force` , it appears as "USBIP Shared Device" without any warning, when I do `usbipd list` it also shows correctly as `Shared (forced)`. The warning appears when it fail to attach.

Thanks for testing. I will try to reproduce it and/or figure out what change caused this (but for now ... I don't know/understand ...).

jtmcdole, 2025-12-06:
Downgrading to 5.2.0 worked for me when forwarding a Heltec Wireless Tacker v1.2 to WSL.

```
usbipd attach --wsl -i 303a:1001                                                                                                                                           ─╯
usbipd: info: Device with hardware-id '303a:1001' found at busid '2-3'.
usbipd: info: Using WSL distribution 'Ubuntu-24.04' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Detected networking mode 'nat'.
usbipd: info: Using IP address 172.31.48.1 to reach the host.
```

```
lsusb
Bus 001 Device 002: ID 303a:1001 Espressif Systems Heltec Wireless Tracker V2
```

```
usbipd list
BUSID  VID:PID    DEVICE                                                        STATE
2-3    303a:1001  USB Serial Device (COM4)                                      Attached
```

Windows Device Manager only shows `USBIP Shared Device` - no hint of forced.

oldgrump-ai, 2026-03-24:
Both 5.2.0 and 5.3.0 DO NOT work for me for an Nvidia Jetson Xavier.

Device shows up as 
`1-1    1d6b:0104  USB Mass Storage Device                                       Shared`

but windows complains and `usbipd attach --wsl` fails to attach