dorssel/usbipd-win #1266: Add policy error

ZZH-Finalize, 2025-09-24:
Error occurs when adding a policy for AutoBind

[image]

dorssel, 2025-09-24:
Yes, good find. I found this out  just a while ago. It was fixed in : https://github.com/dorssel/usbipd-win/commit/37edef316baf003f6b69786295f265fa0ad6cf83. The fix will be in de next release. You could try the current master build, but I'm in a bit of a refactor mode and things may not be stable. ETA about 2 weeks. It is a regression, so could also downgrade to version 5.1.0, which does not have this issue.

dorssel, 2025-09-24:
Additionally: in 5.2.0, policy rules that have *both* a hardware *and* bus ID should work just fine.

dorssel, 2025-09-24:
Additionally (final note): the problem is just in the command line parser of `policy add`. You could downgrade to 5.1.0, then add the policy, and then upgrade to 5.2.0 again.

dorssel, 2025-09-24:
The policy does not attach, it only *allows* attaching. The policy does an auto-bind for a regular user, as if the administrator explicitly  used `bind`.

So instead of (admin) bind + (user) attach, you can do: (admin) policy add + (user) attach. It just allows the administrator to pre-configure certain devices/ports as allowed, without the device actually being present at the time of setting the policy.