dorssel/usbipd-win #1272: Data Security Measures

liRW, 2025-09-27:
How to implement data transmission encryption? Does the official provide a solution? If not, which parts need to be modified to implement it by myself?

liRW, 2025-09-27:
UKEYs are widely used in many application scenarios such as finance and government. If USBIP is used to share UKEYs, is there any way to ensure the security of the certificate transmission within these UKEYs?

dorssel, 2025-09-28:
No, USBIP is inherently "unsafe". It does use regular TCP, so IPSEC could be a solution. This was the original intent of USBIP, see https://elixir.bootlin.com/linux/v6.16.9/source/tools/usb/usbip/src/usbip_network.h#L122. That comment has been there since day one, and was never implemented.

ryangriggs, 2025-11-08:
Wireguard would be an easy solution in this case.

mk-pmb, 2026-08-28:
Modern Windows also has an SSH server that you might try and use as relay.

mk-pmb, 2026-08-28:
> No, USBIP is inherently "unsafe". It does use regular TCP, so IPSEC could be a solution. This was the original intent of USBIP, see https://elixir.bootlin.com/linux/v6.16.9/source/tools/usb/usbip/src/usbip_network.h#L122. That comment has been there since day one, and was never implemented.

That link is now blocked by a bot check. The file seems to be [available on GitHub here](https://github.com/torvalds/linux/blob/master/tools/usb/usbip/src/usbip_network.h#L122); in that version, line 122 currently reads

> /* Negotiate IPSec encryption key. (still not used) */